niceboygithub/AqaraCameraHubfw #2: usage of builtin mosquitto

snakewa, 2021-04-16:
Hi there,

Thank you for guide, I could telnet without any hardware modification or wiring.

Basically I could get snapshot / www cgi script from your works working. I also notice in my firmware, there is running mosquitto. 

It should be handing MQTT. I wonder could I use it to integration with HASS. or could you point out some direction I might work on it. 

I try to connect to MQTT (1883 port) but not working.

I found that you provide mosquitto to Aqaura Gateway (M1 or M1S).

So could I send out some status such as camera motion detection or sound detection or any event by MQTT to HASS?

Thanks,
Tim

niceboygithub, 2021-04-17:
Try this [repo](https://github.com/niceboygithub/AqaraGateway)

But camera motion or sound detection did not pub to mqtt.

gpowa, 2021-04-18:
> 
> 
> Hi there,
> 
> Thank you for guide, I could telnet without any hardware modification or wiring.
> 
Hi,

i try too to telnet without wire and open case, but don`t work. How i can 
activate ist? I`ve tried with FAT32 formated micro sd-card and put a file named "hostname"
with described text. Do i need anything more or a file and folder structure?

gpowa

niceboygithub, 2021-04-19:
try another micro sdcard

fluppie, 2021-05-31:
In my case I needed 2.1.1 to make the HA-plugin work.
https://github.com/mcchas/g2h-camera-mods/issues/11

jstride, 2021-06-16:
Which HA plugin are you using - do you mean the aqara hub one?

niceboygithub, 2022-08-02:
This issue has been open 30 days with no activity

closed 2022-08-02 by niceboygithub