niceboygithub/AqaraCameraHubfw #3: Can't install ota.bin

NikDevx, 2021-04-30:
Hello

What's wrong?
Help me, please.

```
# cd /mnt/sdcard/
# ls
record   log      ota.bin
# cd /etc/
# . ota.sh
firmware version: 2.1.1
firmware build time: 1606803116
payload version: 2.1.1
payload type: 0x20
payload length: 3947667
Segmentation fault
check ota file fail
```
Can't find upgrade file

NikDevx, 2021-04-30:
Or is it possible to run ftp/sambe/netcat/ssh or whatever for copy record from G2H remotely?

niceboygithub, 2021-04-30:
Run /etc/ota.sh not “. ota.sh”

NikDevx, 2021-04-30:
@niceboygithub 
> Run /etc/ota.sh not “. ota.sh”

I tried already  `/etc/ota.sh`
```
# /etc/ota.sh
firmware version: 2.1.1
firmware build time: 1606803116
payload version: 2.1.1
payload type: 0x20
payload length: 3947667
check ota file fail
Can't find upgrade file
```

NikDevx, 2021-04-30:
> Or is it possible to run ftp/samba/netcat/ssh/httpd or whatever for copy record from G2H remotely?

I need to copy files from sdcard remotely
Maybe I can use another one way?

NikDevx, 2021-04-30:
> looks your ota.bin is corrupted.
> try another sdcard.

I tried already 3 sdcard and I bought a new one for G2H

niceboygithub, 2021-05-01:
You need make the space of / (root folder) is enough.  It is required at latest 3MB. 
Also remove any files in /ota

NikDevx, 2021-05-01:
> You need make the space of / (root folder) is enough. It is required at latest 3MB.
> Also remove any files in /ota

You are right!
How I can clear root directory? 

```
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                15.7M     14.0M      1.7M  89% /
tmpfs                    61.2M     48.0K     61.2M   0% /dev
tmpfs                    61.2M         0     61.2M   0% /mnt/usb
tmpfs                    61.2M      8.6M     52.6M  14% /tmp
tmpfs                    61.2M     12.0K     61.2M   0% /var/run
/dev/block/mtdblock3      1.1M    284.0K    804.0K  26% /mnt/config
/dev/block/mtdblock4     10.0M      2.3M      7.7M  23% /customer
/dev/block/mmcblk0p1     29.8G    600.3M     29.2G   2% /mnt/sdcard
# 
```

closed 2021-05-09 by NikDevx