niceboygithub/AqaraCameraHubfw #4: How run RTSP

douglasfl, 2021-05-20:
Hi,
very nice work!

I have a G2 camera down for a while, and I found your work recently and it gave me hope of being able to integrate the camera into the Home-Assistant.

I have access via telnet and TTL for the camera.

I upgraded the camera to its version: G2 ota_3.5.8_0062.bin

My question is how do I run the service rtsp?

Should I copy the repository binaries to the camera's SD, and then execute them?

As I understand it at G2 it would be these commands:
    h264grabber -f &
    rRTSPServer -e &

Again, congratulations on the excellent work!

zhaosk, 2021-06-21:
Hi niceboy, i have the same question with Douglas, but my device is g2h. Really appreciate for your help on running rtsp, looking forward for your advice!

tsunglung, 2021-06-21:
Yes, but not recommend to but execute binary in sdcard

niceboygithub, 2021-06-21:
For G2H, it have more available space of internal storage. I remembered you to put binary in /customer folder.
For G2, it need to be careful to put binary in internal storage.

xztxy, 2021-06-26:
> For G2H, it have more available space of internal storage. I remembered you to put binary in /customer folder.
> For G2, it need to be careful to put binary in internal storage.

G2H  Red light all the time, can't start, can't initialize~what can i do?

niceboygithub, 2021-06-26:
> When I put the binary in, Red light all the time, can't start, can't initialize~what can i do?

Do you have UART log?

xztxy, 2021-06-27:
> > When I put the binary in, Red light all the time, can't start, can't initialize~what can i do?
> 
> Do you have UART log?

I have no log  ~
 I used a programmer All data erased ,can you tell me how to restore？Thanks ！

niceboygithub, 2021-06-27:
> I have no log ~
> I used a programmer All data erased ,can you tell me how to restore？Thanks ！

You need factory image to restore. 
Sorry I did not have factory image.

niceboygithub, 2022-08-02:
This issue has been open 30 days with no activity

closed 2022-08-02 by niceboygithub