niceboygithub/AqaraCameraHubfw #5: Help with telnet

ostgardh, 2021-08-10:
I have try to "create a file 'hostname' which its content as below and put it in sdcard" part
And is not working.

I try to telnet to ip and is not respond.

niceboygithub, 2021-08-12:
What is your version of Aqara Camera?
What is the content of your hostname?

Please save the file 'hostname' as Unix type. You can use Notepad++ to save it.

ostgardh, 2021-08-12:
hostname file.
#!/bin/sh
passwd -d root
echo WITH_TELNET=y >> /etc/.config

Version : [image]

Have try different sdcard 
Have try to chmod the host file to 777

niceboygithub, 2021-08-13:
can you upload the hostname file as attachment

another solution, try [g2h-camera-mods](https://github.com/mcchas/g2h-camera-mods)

ostgardh, 2021-08-14:
[hostname.txt](https://github.com/niceboygithub/AqaraCameraHubfw/files/6986095/hostname.txt)
I have to rename hostname file to hostname.txt

niceboygithub, 2021-08-15:
What is the sdcard format? Is it FAT32?

How do you use telnet? Are your pc and  G2H use the same domain?

ostgardh, 2021-08-15:
Yes. Fat32
Format by camera. 

Have try from my Debian and from putty on windows. 

Same WiFi net yes. 
Possibel to ping. 

I have noticed that the hostname file is deleted after is have be in camera.

eneagiuseppe, 2021-10-02:
Hi @niceboygithub I really appreciated the guide, but I tried to write the hostname (unix format) inside the sdcard (formatted FAT32 or ext4) but it doesn't work. I have last firmware 2.2.3.0001.0526. Can you support us, if is it possible?
Thanks.

niceboygithub, 2021-10-03:
The hostname way still work in 2.2.3

the-gray-ghost, 2021-11-10:
I'm having the same problem with 2.2.3.  Tried two different SDcards Fat32 format.  File saved as 'hostname' no extension in Unix format with Notepad++.  I am using Terminal on Mac to test 'ssh root@192.168.0.113'

ssh: connect to host 192.168.0.113 port 22: Connection refused

niceboygithub, 2021-11-10:
not ssh, please use telnet

ostgardh, 2021-11-12:
@the-gray-ghost
Have you test telnet?
I have upgrade to 2.2.3.

Put hostname file to sdcard with the code.
Same I can´t telnet.

`Trying 192.168.0.160...
telnet: Unable to connect to remote host: Connection refused
# telnet 192.168.0.160
Trying 192.168.0.160...
telnet: Unable to connect to remote host: Connection refused
`

jellejans, 2022-05-20:
@niceboygithub 
@ostgardh 

Is the telnet method still working with 2.2.5 ? because I tried everything, but no luck ?

Kimbo01, 2022-06-14:
would this method works with g2h pro? I tried many times never had any success

MostHated, 2022-06-19:
I have a g2h pro as well and have not had any success. Firmware version is 3.4.0_0008.0004

niceboygithub, 2022-06-20:
The g2h pro did not use hostname method.
G2H pro use [aQRootG3 method](https://github.com/Wh1terat/aQRootG3), but it was invalid in 3.4.0.

closed 2022-08-31 by niceboygithub

isra60, 2022-12-08:
I have v2.2.6 firmware version and it seems that telnet is not getting up. Maybe Aqara has removed it in that version? The model is G2H.

The other method xiaomi mi home soft_hack. Is there any tutorial ?