niceboygithub/AqaraCameraHubfw #8: Any Guide to Flash G2 Aqara

vlar, 2021-09-05:
I apologise for the question if obvious but how does one get started to flash a G2 aqara to get a RSTP stream ? 

I have hope to finally get a use for my G2 camera. .

niceboygithub, 2021-09-06:
There is no custom or stock firmware for g2.

to enable rtsp.
1. Enable telnet to copy execute binaries to g2
2. Copy h264grabber and rRTSPServer to g2
3. Run theses execute binary. Also you can config them running while g2 boot up.

meeDamian, 2022-01-17:
How to enable telnet on G2, thought? I've spent last two hours trying to find a way, and all I'm able to find is either guides for G2H, or links back to **your** repos, which unfortunately, don't really answer this question :(.

Would you happen to have any links/guides/tips on hand?

I really want to get it to work. Especially now, that the official app got borked, and I'm no longer able to even open the camera or its settings :( .

https://user-images.githubusercontent.com/526657/149741730-3b70fed0-6d72-4514-b05b-215f0b021bfc.mp4

niceboygithub, 2022-01-18:
> How to enable telnet on G2, thought? I've spent last two hours trying to find a way, and all I'm able to find is either guides for G2H, or links back to **your** repos, which unfortunately, don't really answer this question :(.

did you try the way of enable telnet on [G2H](https://github.com/mcchas/g2h-camera-mods)?

I enabled G2 telnet by solder out UART tx/rx. Please see[ item 7](https://github.com/niceboygithub/AqaraCameraHubfw/tree/main/binutils)

> Would you happen to have any links/guides/tips on hand?
> 
> I really want to get it to work. Especially now, that the official app got borked, and I'm no longer able to even open the camera or its settings :( .
> 
>  screen-20220117-024936_2.mp4

Try to clean user data of app

meeDamian, 2022-01-27:
> did you try the way of enable telnet on G2H?

It did not work for me, either because it's G2 or the firmware it's running.

> I enabled G2 telnet by solder out UART tx/rx. Please see item 7

I've soldered the cables, and connected it to my RBP zero, but whenever I try to `screen /dev/ttyS0 115200`, ` minicom -b 115200 -o -D /dev/ttyS0`, `tio -b 115200 /dev/ttyS0`, etc - all I get back is just gibberish 😅 - apologies if my question is as basic, but despite my many attempts to figure it out, this step has so far defeated me :/. So… any hints highly appreciated!

[image]

niceboygithub, 2022-01-28:
Try 38400

I already forgot the baud rate

closed 2022-08-31 by niceboygithub