niceboygithub/AqaraCameraHubfw #9: G2H does not start

geldelo, 2021-11-09:
Hello. Foolishly deleted the /tmp/out/ folder and the /etc/normal.xml file (rooted G2H). The yellow indicator is now on all the time, it does not appear on the network. What can I do?

niceboygithub, 2021-11-09:
You need use UART to create /etc/normal.xml

```
<monitor>
        <process name="/local/bin/property_service">
                <arg desc="loglevel option">-i</arg>
                <arg desc="loglevel 0">/etc/build.prop</arg>
                <arg desc="data path option">-p</arg>
                <arg desc="data path">/mnt/config/prop.dat</arg>
        </process>
        <process name="/tmp/out/mosquitto">
                <arg desc="foreground run"></arg>
        </process>
        <process name="/tmp/out/ha_agent">
                <arg desc="foreground run"></arg>
        </process>
        <process name="/tmp/out/proxy_server">
                <arg desc="foreground run">-D</arg>
        </process>
        <process name="/tmp/out/camera">
                <arg desc="bring up by monitor">-M</arg>
        </process>
        <process name="/tmp/out/ha_master">
        <arg desc="bring up by monitor">-a</arg>
        <arg desc="bring up by monitor">/lib/libha_auto.so</arg>
        <arg desc="bring up by monitor">-g</arg>
        <arg desc="bring up by monitor">/lib/libha_energy.so</arg>
        <arg desc="bring up by monitor">-G</arg>
        <arg desc="bring up by monitor">/mnt/config</arg>
        </process>
        <process name="/tmp/out/zigbee_agent">
        <arg desc="bring up by monitor">-f</arg>
        <arg desc="bring up by monitor">/etc/zigbeeAgent.conf</arg>
        </process>
        <process name="/tmp/out/ha_driven">
        <arg desc="bring up by monitor">-d</arg>
        <arg desc="bring up by monitor">/mnt/config/ha_driven</arg>
        </process>
</monitor>
```

geldelo, 2021-11-09:
Thx. I do not know how to do that

niceboygithub, 2021-11-10:
Uart connection Picture is [here](https://github.com/niceboygithub/AqaraCameraHubfw/blob/main/images/ttl_uart.jpg)

The use putty to open COM port (38400,8n1)

If UART connection is ok, you can enter 'vi' command in putty
```
vi /etc/normal.xml
```
Next enter 'i', then copy&paste. After pasted, enter 'ESC' then ':' , "wq" enter..

geldelo, 2021-11-10:
Thx. I saw this picture. I ask you to excuse me, I am a newbie in such matters.

In the picture I see 2 pins RX and TX. But I understand that 2 more contacts are needed: GND and 3.3v, but I don't see them in the picture, I can't find them myself. Or I'm wrong?

geldelo, 2021-11-10:
Update. I disassembled the camera and found a board with all four necessary pins for connection. Can I solder to these contacts?

[image]

niceboygithub, 2021-11-10:
> Update. I disassembled the camera and found a board with all four necessary pins for connection. Can I solder to these contacts?
> 

I did not test it before. You have a try.

geldelo, 2021-11-10:
Thx. I'll write after the tests

geldelo, 2021-11-16:
[image]

I soldered the pins to the correct connectors and connected to the Uart device.

When connected to a device via Putty, I used 38400.8n1.

But Putty won't connect. I do not know what to do. Help me please.

niceboygithub, 2021-11-17:
Please use the TX, RX in https://github.com/niceboygithub/AqaraCameraHubfw/blob/main/images/ttl_uart.jpg

geldelo, 2021-11-18:
> Please use the TX, RX in https://github.com/niceboygithub/AqaraCameraHubfw/blob/main/images/ttl_uart.jpg

I soldered the contacts according to your scheme, but unfortunately Putty did not connect.

niceboygithub, 2021-11-19:
How did the TX, RX connection?

the correct way is
G2H RX <-> TX of USB2Uart
G2H Tx <-> TX of USB2Uart

geldelo, 2021-11-20:
Thx, i connected. But can't etner any command.
On connect error:

19700101 03:00:34.249 ERROR    monitor 115: get RootElement fail
19700101 03:00:34.250 ERROR    monitor 315: parser_processlist[/etc/normal.xml] fail
[RCS]: /etc/init.d/S91telnetd

niceboygithub, 2021-11-21:
G2H TX is ok now
But G2H RX is still not connected well.
Please check by volt-ohm-milliammeter

rezmus, 2021-11-22:
you can try to fix with factory.ini on sdcard. check /etc/factory.ini for example. you should be able to bring network back and fix issue via telnet.

geldelo, 2021-11-23:
> you can try to fix with factory.ini on sdcard. check /etc/factory.ini for example. you should be able to bring network back and fix issue via telnet.

Thanks for the help. But I need more information. I'm afraid, I can't do it myself. Could you help me with the correct step-by-step plan and command writing?

geldelo, 2021-11-25:
So, i have not been able to reanimate G2H using Uart (read, but does not allow entering commands).

Can you tell me, if it is possible to create and write "normal.xml" using "hostname" file from SD card?

niceboygithub, 2021-11-25:
> So, i have not been able to reanimate G2H using Uart (read, but does not allow entering commands).
> 
> Can you tell me, if it is possible to create and write "normal.xml" using "hostname" file from SD card?

yes, it is good idea.

geldelo, 2021-12-09:
> > So, i have not been able to reanimate G2H using Uart (read, but does not allow entering commands).
> > Can you tell me, if it is possible to create and write "normal.xml" using "hostname" file from SD card?
> 
> yes, it is a good idea.

And so it didn’t work

rezmus, 2021-12-09:
because hostname is probably called from one of binaries that wont start due to broken normal.xml ;d

geldelo, 2021-12-09:
> you can try to fix with factory.ini on sdcard. check /etc/factory.ini for example. you should be able to bring network back and fix issue via telnet.

Can you help me with this?

rezmus, 2021-12-09:
put on sdcard empty factory.ini and hostname like below (make sure it has linux line endings). check uart log during boot.

```
#!/bin/sh

rm -rf /mnt/sdcard/factory.ini
passwd -d root
echo WITH_TELNET=y >> /etc/.config
cp -f /etc/factory.xml /etc/normal.xml

echo "hostname fix"
```

geldelo, 2021-12-10:
> 

Thanks, but that didn't help either.

rezmus, 2021-12-10:
uart log when you boot with factory.ini and hostname on sdcard?

geldelo, 2021-12-10:
> uart log when you boot with factory.ini and hostname on sdcard?

I did it without UART. There is no possibility with Uart yet.

rezmus, 2021-12-10:
you can't really do this blind... try this factory.ini (remember unix line endings). change ssid/key/gateway/netmask to your wifi router data and set ip to something not used by any device connected to router. camera should connect to this wifi network, try to ping or telnet in.

```
[factory]
mode = 4

[wifi]
ssid = rdoffice
key = 12345678

[network]
dhcp = off
ip = 192.168.1.88
netmask = 255.255.255.0
gateway = 192.168.1.1

[cut]
peroid = 15
```

geldelo, 2021-12-10:
Wow. Thx, it ping's!

geldelo, 2021-12-10:
> you can't really do this blind... try this factory.ini (remember unix line endings). change ssid/key/gateway/netmask to your wifi router data and set ip to something not used by any device connected [...]

But I can't do anything further. Putty can't connect.

geldelo, 2021-12-10:
> you can't really do this blind... try this factory.ini (remember unix line endings). change ssid/key/gateway/netmask to your wifi router data and set ip to something not used by any device connected to router. camera should connect to this wifi network, try to ping or telnet in.

Now i can connect via telnet, but my previous password to login is incorrect.
Can you please tell me what to do next to make the device to factory settings? 
If I take out the SD card, the device does not connect to Wi-Fi.

rezmus, 2021-12-10:
try root and no password (enter).

geldelo, 2021-12-10:
> 

Login incorrect

geldelo, 2021-12-10:
> try root and no password (enter).

Login incorrect

geldelo, 2021-12-10:
> try root and no password (enter).

While we are looking for ways to reanimation of this device, I would like to ask you: How did you know about the factory.ini? No one before it knew about this file and what he was doing. Are you a developer of Aqara :)?

rezmus, 2021-12-10:
try root 09qjuS@3

if does not work put hostname file on sdcard

```
#!/bin/sh

echo "root:mypass123" | chpasswd
```

boot camera with sdcard, wait 1 minute, reboot camera, wait 1 minute and try to login root mypass123

geldelo, 2021-12-10:
> try root 09qjuS@3
> 
> if does not work put hostname file on sdcard
> 
> ```
> #!/bin/sh
> 
> echo "root:mypass123" | chpasswd
> ```
> 
> boot camera with sdcard, wait 1 minute, reboot camera, wait 1 minute and try to login with root mypass123

I did everything according to the instructions, but I can't connect. 

Camera-Hub-G2H login: root
Password: 
Login incorrect

geldelo, 2021-12-10:
> try root 09qjuS@3
> 
> if does not work put hostname file on sdcard
> 
> ```
> #!/bin/sh
> 
> echo "root:mypass123" | chpasswd
> ```
> 
> boot camera with sdcard, wait 1 minute, reboot camera, wait 1 minute and try to login with root mypass123

With your help, I managed to revive the camera. How can I thank you?

rezmus, 2021-12-10:
run this cmd in telnet. it should fix broken normal.xml file when you boot without factory.ini

```
cp /etc/factory.xml /etc/normal.xml
```

there was a sample /etc/factory.ini in firmware. all you need to do is check how it works.

geldelo, 2021-12-10:
> run this cmd in telnet. it should fix broken normal.xml file when you boot without factory.ini
> 
> ```
> cp /etc/factory.xml /etc/normal.xml
> ```
> 
> there was a sample /etc/factory.ini in firmware. all you need to do is check how it works.

How can I thank you?

ashwindz, 2022-01-09:
Hello @rezmus @niceboygithub @geldelo  - I got this camera about a month back. It was paired to Homekit and working fine. About 2 weeks back, this camera dropped off the network. I could not bring it back online so I removed it from homekit hoping I could pair it again.

But, try as I might with multiple resets, it just looping through "ready to connect. please open the aqara home app" and does not get paired with Homekit. Same goes with the Aqara app - it just wont add the camera  because QR scan is not completed.

I see that @geldelo was able to revive the camera. Is there any way to use the SD card to restore this? Please advice! 

PS: Wifi is 2Ghz only and has no underscores.

rezmus, 2022-01-09:
try factory reset (click button 10 times).

ashwindz, 2022-01-09:
> try factory reset (click button 10 times).

Thanks @rezmus for the suggestion. Will Try it. 

By Button do you mean the button on top? or the reset button? How will I knoww the reset was successful ?

rezmus, 2022-01-09:
top, 10 fast clicks. it will say it does factory reset.

ashwindz, 2022-01-09:
Thanks but no luck @rezmus - I got the message that reset was successful but it still wont pair on HomeKit. Did restart my iPhone too. Anything else I can try?

rezmus, 2022-01-09:
try to pair in aqara app. if it still fails, you should probably write to aqara customer support. they handle such cases.

ashwindz, 2022-01-09:
Ok - I did mail them a video and some other info they wanted. Hope to hear from them. thanks again @rezmus

ashwindz, 2022-01-17:
@rezmus - Aqara wants me to return the camera for a replacement and has no suggestions. Unfortunately I got this for use in India and wont have someone in the US for a while. 

I just tried hostname with 
`cp -f /var/log/* /mnt/sdcard/`

Nothing came up after boot. I guess whatever loads hostname is broken :(

evripidis, 2023-01-18:
Is it possible to load the original files on the SD card are reverse all those actions? Does anyone have a backup to share?

My camera just stays with a blue light, it is not reseting, not connecting to network tried all the non-UART tips above :-(

Topereb, 2023-02-18:
Thank you very much for your work.
Please tell me there is an Aqara g2h camera (Chinese region) after an unsuccessful update, the camera hangs on a red fire, the reset does not help.
Can it be restored somehow?
There is a dump from this non-working camera, and a dump from a working but European region.
I uploaded a dump from the working one to the non-working camera, it started up but did not connect to the Akara home and was an exact copy of the working camera, the poppy address and device id were the same. Both refused to work on the same network.
Maybe someone can help me with this, not for free of course.
Can I edit the dump in a hex editor?
Who can help with this
Thank you 🙏🏻

Topereb, 2023-02-18:
> > So, i have not been able to reanimate G2H using Uart (read, but does not allow entering commands).
> > Can you tell me, if it is possible to create and write "normal.xml" using "hostname" file from SD card?
> 
> yes, it is a good idea.

Thank you very much for your work.
Please tell me there is an Aqara g2h camera (Chinese region) after an unsuccessful update, the camera hangs on a red fire, the reset does not help.
Can it be restored somehow?
There is a dump from this non-working camera, and a dump from a working but European region.
I uploaded a dump from the working one to the non-working camera, it started up but did not connect to the Akara home and was an exact copy of the working camera, the poppy address and device id were the same. Both refused to work on the same network.
Maybe someone can help me with this, not for free of course.
Can I edit the dump in a hex editor?
Who can help with this
Thank you 🙏🏻

c20h32o5, 2023-03-22:
> > > So, i have not been able to reanimate G2H using Uart (read, but does not allow entering commands).
> > > Can you tell me, if it is possible to create and write "normal.xml" using "hostname" file from SD card?
> > 
> > 
> > yes, it is a good idea.
> 
> Thank you very much for your work. Please tell me there is an Aqara g2h camera (Chinese region) after an unsuccessful update, the camera hangs on a red fire, the reset does not help. Can it be restored somehow? There is a dump from this non-working camera, and a dump from a working but European region. I uploaded a dump from the working one to the non-working camera, it started up but did not connect to the Akara home and was an exact copy of the working camera, the poppy address and device id were the same. Both refused to work on the same network. Maybe someone can help me with this. Can I edit the dump in a hex editor? Who can help with this Thank you 🙏🏻

Hi. I need dump rootfs.bin European region version 2.2.7. Help me?

almirus, 2023-08-27:
@rezmus 
Hello. My camera also died after editing normal.xml. Is it possible to revive the camera without opening it? Can it be flashed via an SD card?

closed 2024-01-25 by niceboygithub

redelva, 2024-03-27:
[image]
 when i enter screen /dev/cu.xxx 115200, it enter screen and no output, any sugeestions?