niceboygithub/AqaraCameraHubfw #10: G2H stuck in a loop

nedisonn, 2022-01-12:
I've accidentally overwritten the S90app file with the hostname file. I tried getting access through UART but no luck, I can read the logs but can't write into it. It loads fine until it loads the S90app script and it gets stuck in the loop. Is there a way I can access this file and revert what I did?

nedisonn, 2022-01-14:
It literally has the normal log, I overwrote the S90app with a code that had a loop, so when it starts that it gets stuck, I would need to find a way to reoverwrite that file but I can't seem to be able to, any ideas?

rezmus, 2022-01-14:
look at the begining of the log. looks like uboot can recover rootfs from sdcard.

nedisonn, 2022-01-14:
How would I do that? :)

rezmus, 2022-01-14:
ask someone for rootfs.bin dump, put on sdcard, check for errors.

nedisonn, 2022-01-14:
Do you know where is the rootfs.bin located? I have another G2H I can try to get this dump... But I can't find the .bin 😄

rezmus, 2022-01-14:
download full busybox and put on sdcard

https://busybox.net/downloads/binaries/1.31.0-defconfig-multiarch-musl/busybox-armv7l

```
/mnt/sdcard/busybox-armv7l dd if=/dev/mtd/mtd6 of=/mnt/sdcard/rootfs.bin
sync
```

it will dump rootfs from working camera to rootfs.bin on sdcard. try to boot broken camera with this card and check uart log.

nedisonn, 2022-01-14:
Thanks! But no luck, it's in the same as before, then it hardware resets....

IPL gd156225
D-06
HW Reset
128MB
BIST0_0001-OK
offset:00010000
size:7fc8 chks:5551a134 ok

IPL_CUST gbf16da4
MXP found at 0x00020000
  decomp_size=0x0003e268

U-Boot 2015.01 (Jun 19 2020 - 15:49:12)

Version: I3g2e8b5bf
DEVINFO: NON
[WDT] Enalbe WATCHDOG 60s
       Watchdog enabled
I2C:   ready
DRAM:  64 MiB
WARNING: Caches not enabled
MMC:   MStar SD/MMC: 0
nor_flash_mxp allocated success!!
Flash is detected (0x0513, 0xC2, 0x20, 0x19)
SF: Detected nor0 with total size 32 MiB
MXP found at mxp_offset[1]=0x00020000, size=0x1000
env_offset=0x4F000 env_size=0x1000
Flash is detected (0x0513, 0xC2, 0x20, 0x19)
SF: Detected nor0 with total size 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
gpio[96] is 1
gpio[97] is 0
Net:   Net Initialization Skipped
No ethernet found.
get file[mtd.bin] fail
get file[uboot.bin] fail
get file[kernel.bin] fail
get file[kernel2.bin] fail
get file[customer.bin] fail
load file[rootfs.bin]
reading rootfs.bin

niceboygithub, 2022-01-15:
The dumped raw rootfs binary may different from the uboot.bin that uboot flashing.

Currently, I can not find the kernel.bin/rootfs.bin for uboot flashing after I did the google search.

Back to the stuck problem.
Your S90app is corrupt. So G2H did not continue to boot up. 

There is high risk way may help to recovery. But I am not recommend to normal user to do it.
1. Stop uboot to boot before loading kernel. To stop uboot, need to interrupt flash reading.
2. change bootcfg "init=/linuxrc" to "init=/bin/sh"
3. edit /etc/init.d/S90app

The best solution, call Aqara Service.

nedisonn, 2022-01-17:
I'm not a normal user, I'm just not used to this way of doing it. How exactly would I interrupt the flash reading? Thanks!

niceboygithub, 2022-01-18:
You can find the pinout of flash IC and short data pin to ground while booting up kernel.

nedisonn, 2022-01-19:
Is there a right moment to do this? I tried and it stops altogether, and it doesn't let me interact with the console also...
I tried calling Aqara service and they said there's nothing they can do to help me. :(

almirus, 2023-08-27:
i have same problem, same log

closed 2024-01-25 by niceboygithub

redelva, 2024-02-26:
any guide for uart of g2h? my g2h stuck at boot, need to stock firmware

almirus, 2024-02-26:
@redelva I restored the firmware of the camera using ch341 and a clip, directly flashing the patched image downloaded from the camera. The error was in the configuration.

redelva, 2024-02-26:
in my option, restore to stock firmware means camera will start from scratch.

could u offer some picture of connect to camera and commands for ch341?

does it work on mac, no windows available now

almirus, 2024-02-26:
@redelva 
[image]
[image]

redelva, 2024-02-26:
i am a newbie for hardware connection. 

Just found some ch341 from ali express, found that it support uart and i2c mode.

for uart mode, it use gnd, tx, rx to connect. 

just found one connect point from redline clip on above image.

redelva, 2024-02-26:
just found ch341 and clip, is it ok to restore stock firmware via uart?

for modified firmware, it can not boot up?

1.Stop uboot to boot before loading kernel. To stop uboot, need to interrupt flash reading.
2.change bootcfg "init=/linuxrc" to "init=/bin/sh" then continue to boot kernel.
3.After boot to sh, edit /etc/init.d/S90app

do u have any idea to do 3 steps? it will be easy for me when change bootcfg, cause i update /etc/init.d/S90app with httpd, it stopped whole boot up

almirus, 2024-02-26:
@redelva I couldn't recover the firmware via UART, I didn't use it. I used SPI. The red wire on the clamp should be positioned where I indicated.
 I had a problem due to incorrect camera configuration, which caused it to go into a bootloop. I connected the clamp, dumped the firmware to the computer, opened the firmware file in a HEX editor, corrected the incorrect configuration (in a text file), and flashed the firmware back using the clamp. This allowed the camera to start up again. I couldn't interrupt the bootloop via UART.

redelva, 2024-02-26:
it should be same case for me. what i wanna do is update init script and remove httpd.

which os r u using? hope it is mac

almirus, 2024-02-26:
@redelva win11
I believe the operating system won't be an issue, as there are tools available for both MAC and Windows.
for example https://github.com/McMCCRU/SNANDer/tree/main/MacOSX