niceboygithub/AqaraCameraHubfw #13: Cannot connect to G2H telnet

a000a498, 2022-07-13:
When my UART is plugged into the computer, the camera doesn't respond, neither does the telnet
please help me to see where I am doing wrong
[image]
[image]
[image]

niceboygithub, 2022-07-14:
You need swap RX and TX, Also do not connect 5V from pcb.
Use USB power cable to power up G2H.

a000a498, 2022-07-14:
> You need swap RX and TX, Also do not connect 5V from pcb.
> 
> Use USB power cable to power up G2H.

So GND still connect right?

a000a498, 2022-07-14:
[image]
[image]
I swapd RX and TX and remove 5V cable also use USB cable to power up then still cannot connect :(

a000a498, 2022-07-14:
Update, I have successfully connected but the root account and password I entered are incorrect

baobaobuku1210, 2022-07-28:
Why can't I enter the user name and password
[image]

a000a498, 2022-07-30:
I open telnet now,but why i can not connect to aqara home :(
before i had plug sd card in to camera then it cannot connect to app

[image]
[image]

baobaobuku1210, 2022-07-30:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

closed 2023-11-22 by niceboygithub

redelva, 2024-03-27:
[image]

when enter screen /dev/cu.xxxx 115200

it enter screen and no output, what should i do now?