niceboygithub/AqaraCameraHubfw #14: RTSP script autoload not working

NeroWard, 2022-08-04:
I was able to run the RTSP server in high quality on G2 firmware 3.5.8_0062, but there was a problem with RTSP starting automatically. If you try to manually run the rtsp-start.sh script, the telnet command line hangs, only rebooting the camera helps. Question: how to force the rtsp server to start?
 P/S sorry for my english

niceboygithub, 2022-08-05:
The size of memory in G2 is too small.
It causes rtsp can’t run with other processes.

NeroWard, 2022-08-06:
I ran rtsp manually, telnet worked and g2 responded to commands

NeroWard, 2022-08-10:
There is also a display issue. Possible solutions?
In Home Assistant, the stream is received but with similar problems, there are still problems with playing the video stream in Home Assistant
[image]

NeroWard, 2022-08-11:
The image "blinks" and a normal image appears for a while, artifacts appear later

GeorgyBenelli, 2022-09-23:
> The image "blinks" and a normal image appears for a while, artifacts appear later

I have the same problem. Did you solve the video quality problem?

NeroWard, 2022-09-26:
> > The image "blinks" and a normal image appears for a while, artifacts appear later
> 
> I have the same problem. Did you solve the video quality problem?
I deleted the "Log" folder on the SD drive, it didn't help at all, and then it stopped helping to fix the problem at all

creampenne, 2022-12-12:
Did you solve the problem?

NeroWard, 2022-12-13:
> Did you solve the problem?

No, the problem has not been resolved. A bug occurs and I don't know what to do about it

closed 2023-11-22 by niceboygithub