niceboygithub/AqaraCameraHubfw #15: G3 failed to rotate or pan after downgrading  to 3.3.4

pleaseRedo, 2022-08-11:
Hi, thanks for your fw repo.

I manage to downgrad my G3 from 3.4.1 to 3.3.4, for the purpose of enabling telnet. Eventually, I integrated this downgraded G3 into ha but realized there is no way to rotate the camera at all (using aqara home app to rotate the camera.). 

Is this phenomenon normal?

niceboygithub, 2022-08-13:
Were You already downgraded the G3?

The PTZ of G3 from Aqara Home is slow because the command is come from cloud.

in my experience, the G3 sometimes is not responded to PTZ command.

pleaseRedo, 2022-08-13:
Thanks for your reply,

I already downgraded the G3, Aqara Home App also diplayed the downgraded version number.  

I think G3 faild to respond to those PTZ commands, here is what I discovered:

- Can't rotate, I pressed directional button and wait, nothing happends.
- G3 do reacts to sleep command, it flip and showed sd card slot. 
- G3 **partially** reacts to wake command, it won't flip and stucks at sleep mode orientation(sd card slot). But G3's camera is active, I can see the video feed, but blurred out.

-  Upgrading to latest version via Aqara Home App fixes aboves.

rezmus, 2022-08-13:
did you only downgrade rootfs?

pleaseRedo, 2022-08-15:
> 
Yes, only rootfs. Would you mind telling me what else needs to be downgraded? 

Thanks.

rezmus, 2022-08-15:
do you know factory firmware it came with?

pleaseRedo, 2022-08-16:
> 

I only got stock firmware from this repo `https://github.com/niceboygithub/AqaraCameraHubfw/tree/main/stock/G3`

I then extract rootfs from fw.bin to downgrade.

I'm not really sure if this stock firmware is what you referred **factory fw**.

rafaelreis-r, 2022-08-17:
Hey! I would also be interested on downgrading my G3. A bit lost though, since I'm not familiar with firmware bin files manipulation. @pleaseRedo what procedure did you follow?

pleaseRedo, 2022-08-18:
> Hey! I would also be interested on downgrading my G3. A bit lost though, since I'm not familiar with firmware bin files manipulation. @pleaseRedo what procedure did you follow?

1. Get 'rootfs.bin' of firmware version 3.3.4. (use 7z to extract the firmware bin)
2. Copy 'rootfs.bin' to sdcard which is using FAT32 format.
3. Power Off G3.
4. Insert the sdcard to G3.
5. Press the front button of G3.
6. Power on G3.
7. Wait three seconds and release the button.
8. If the LED is turned to RED, it starts to flash 'rootfs.bin'.

I followed these steps. Downgraded, but encoutered above problems. Would like to know how you would ended it up with.

rafaelreis-r, 2022-08-19:
Hey @pleaseRedo 

Thank you for the information. Just tried it. 

Went from 3.4.1 straight to 3.2.7 instead of 3.3.4; Light turns purple on mine, not red.

I also have the same issues you reported. No PTZ, but also No Wake / sleep function. I remember reading 3.4.1 release notes, and it mentions improvement to PTZ smoothness. I'm sure they changed something else besides rootfs.bin which may have caused our issues. 

After that i tried rooting it anyways https://github.com/Wh1terat/aQRootG3

I followed the instrucions, it read the QR code with the payload. Then when I restarted it to add it normally to the Aqara app, it stopped responding. It powers on (green light), does a PTZ spin and turns off all lights. Tried rebooting it multiple times, holding the button, pressing the button 10+ times, it simply stopped responding. 

I might have bricked mine

Hope @rezmus could shed some light on where to poke around.

niceboygithub, 2022-08-20:
I have no answer why the PTZ of your G3 does not work well. But I recommended to use 3.3.9 firmware

pleaseRedo, 2022-08-22:
> 

> I followed the instructions, it read the QR code with the payload. Then, when I restarted it to add it normally to the Aqara app, it stopped responding. It powers on (green light), does a PTZ spin and turns off all lights. Tried rebooting it multiple times, holding the button, pressing the button 10+ times, it simply stopped responding.

Have you tried other version of 'rootfs.bin' ? Could possibly unbrick your G3.

Here is what I did after downgrade:
Right after downgradding, I add G3 to Aqara app without root, I did this cuz I'd like to verify the version. Then I successfully root it and integrate it into my HA. But no PTZ. 

During downgrade, light turns in pink, definitely not purple in your vid.

rafaelreis-r, 2022-08-22:
Hey @pleaseRedo thank you for the hint. 

Rolling back to 3.4.1 rootfs recovers the camera. It works normally, albeit non-rooted. 
As soon as I downgrade to 3.3.4 (or any other version for that matter) it goes back into the unresponsive pattern shown in the video. 

It looks like I'm locked to 3.4.1 fw or above. 
I will try to experiment further a little. But it seems we might be out of luck.

Couldn't the rootfs / fw for the G3 be modded, as done to other hubs?

rezmus, 2022-08-22:
it is signed. you can try to downgrade kernel linux.bin too. do you know fw version your camera came with? or production date from sticker.

rafaelreis-r, 2022-08-22:
Hey @pleaseRedo, 

After poking a little more, I was able to get to the same state as yours. Rooted non PTZ. 

All I had to do was **reset camera do default settings by pressing button 10 times BEFORE downgrading.** If you downgrade with pre-set settings / bound to the Aqara app it will bork it, as I showed in the video.

Maybe this detail could be added to the main Readme.md, since it is important.

@rezmus I don't have the manufacturing date. I do recall it came with a very early firmware, probably 3.2.7 or 3.2.4, since it had to update a few times to get to the latest fw. I have telnet access now on 3.3.4, let me know if I can provide any more information.

rezmus, 2022-08-22:
can you try 3.3.9? this version can keep root with locked post_init.sh script.

rafaelreis-r, 2022-08-22:
Hey @rezmus just to be clear before I do anything. After downgrade to 3.3.4 + root, I did this:

``` 
chmod a+w /data/scripts/post_init.sh
echo -e "#!/bin/sh\n\nasetprop sys.camera_ptz_moving true\nfw_manager.sh -r\nfw_manager.sh -t -k" > /data/scripts/post_init.sh 
chattr +i post_init.sh 
```

Also did this:

``` 
mkdir -p /data/ota_dir
touch /data/ota_dir/lumi_fw.tar
chattr +i /data/ota_dir/lumi_fw.tar 
```

> can you try 3.3.9? this version can keep root with locked post_init.sh script.

You are asking me to:
1 - reset camera (10 button presses)
2 - upgrade rootfs.bin to 3.3.9
3 - see if root is retained

Is that it?

rezmus, 2022-08-22:
yes

rafaelreis-r, 2022-08-24:
@rezmus @pleaseRedo 

On 3.3.9 PTZ and Sleep / Wake works again. Human tracking etc works as well. 

Root is not retained. Telnet connection is gone:

```
telnet 192.168.1.144
Trying 192.168.1.144...
telnet: connect to address 192.168.1.144: Connection refused
telnet: Unable to connect to remote host
```

rezmus, 2022-08-24:
are you 100% sure post_init.sh was locked on 3.3.4? because cmd you pasted miss full path. you can root on 3.3.4 and confirm post_init.sh has **i** flag with
```
lsattr /data/scripts/post_init.sh
```
root should still be there after 3.3.9 update.

rafaelreis-r, 2022-08-24:
@rezmus Yes, I did notice the missing path in the Readme.md documentation and used full path when entering the command. I'm 100% sure.

rezmus, 2022-08-24:
can you try again, this time without factory reset between 3.3.4 > 3.3.9 update? i checked 3.3.9 and it should still call post_init.sh so unless they added some stuff to kill telnetd later it should work fine.

rafaelreis-r, 2022-08-25:
> can you try again, this time without factory reset between 3.3.4 > 3.3.9 update? i checked 3.3.9 and it should still call post_init.sh so unless they added some stuff to kill telnetd later it should work fine.

SUCCESS!

3.3.4 rooted:

```
telnet 192.168.1.144
Trying 192.168.1.144...
Connected to camera-hub-g3-b417.lan.
Escape character is '^]'.

Camera-Hub-G3-B417 login: root
1 ulimit=256
~ # uname -a
Linux Camera-Hub-G3-B417 4.9.84 #35 SMP PREEMPT Thu Dec 16 18:30:56 CST 2021 armv7l GNU/Linux
~ # chmod a+w /data/scripts/post_init.sh
~ # echo -e "#!/bin/sh\n\nasetprop sys.camera_ptz_moving true\nfw_manager.sh -r\
nfw_manager.sh -t -k" > /data/scripts/post_init.sh
~ # chattr +i /data/scripts/post_init.sh
~ # reboot now
~ # ps a | grep tt
 1315 root      0:00 mosquitto -d
 1375 root      0:03 Z3GatewayHost_MQTT -n 1 -b 115200 -p /dev/ttyS1 -d /data/
 1798 root      0:00 grep tt
~ # lsattr /data/scripts/post_init.sh
-----i---c--- /data/scripts/post_init.sh
```

Upgrade to 3.3.9 only rootfs.bin without factory reset:

```
 telnet 192.168.1.144
Trying 192.168.1.144...
Connected to camera-hub-g3-b417.lan.
Escape character is '^]'.

Camera-Hub-G3-B417 login: root
1 ulimit=256
~ # uname -a
Linux Camera-Hub-G3-B417 4.9.84 #35 SMP PREEMPT Thu Dec 16 18:30:56 CST 2021 armv7l GNU/Linux
~ # ps a | grep tt
 1332 root      0:00 mosquitto -d
 1351 root      0:02 Z3GatewayHost_MQTT -n 1 -b 115200 -p /dev/ttyS1 -d /data/ -r c
 1556 root      0:00 grep tt
~ # lsattr /data/scripts/post_init.sh
-----i---c--- /data/scripts/post_init.sh
```

PTZ, Human tracking works, etc works!

rafaelreis-r, 2022-08-25:
@rezmus this is a side quest, I didn't want to open a new isse just for it, plz let me know if you want me to open a separate one just in case. 

I've been working with Zigbee2MQTT devs to support Aqara A100 locks. (you can have a look [here](https://github.com/Koenkk/zigbee2mqtt/issues/13087))
We got to a point where the zigbee communication from G3 hub to A100 lock is encrypted. We cannot reverse engineer it. We were wondering if you know where you can find that key. 

I just dumped a bunch of zigbee info over telnet, but no luck:
[zigbee_info_dump.txt](https://github.com/niceboygithub/AqaraCameraHubfw/files/9425952/zigbee_info_dump.txt)

I noticed ControlBridge.bin is the coordinator fw. 

Have you ever been able to extract Coordinator fw for any of the Aqara hubs? It doesn't have to be G3 one, any would work.

rezmus, 2022-08-25:
can't really help with zigbee, but check ota.bin you extracted to get rootfs for downgrade/update.

closed 2022-08-26 by pleaseRedo