niceboygithub/AqaraCameraHubfw #16: RTSP stream protection

NeroWard, 2022-08-20:
Is it possible to create a login and password to protect an RTSP stream?How can I do that?

niceboygithub, 2022-08-21:
Do you mean telnet login or rtsp password?

To change login password, you use the below command to change password after telnet to.
```
passwd
```

To change password of rtsp stream, you can use the below command but the username and password are dynamic.
```
killall rtsp; rtsp -a
```

NeroWard, 2022-08-22:
I mean the RTSP password. Another question, how to find out the RTSP login and password using the command? Is it possible to make a static RTSP login and password?

niceboygithub, 2022-08-26:
As I know, it uses dynamic login/password and can not change the usage.

to get login/password.
```
agetprop sys.camera_rtsp_url
```

NeroWard, 2022-09-04:
Doesn't work, says command not found. Could you update rRTSPServer to the latest version of the hack
yi-hack-MStar?

niceboygithub, 2022-09-04:
Above command are for Aqara Camera G3

What is your camera model?

NeroWard, 2022-09-04:
G2

closed 2023-11-22 by niceboygithub