niceboygithub/AqaraCameraHubfw #23: Enable Telnet of Aqara Gateway M2 2022 link fail

tomtzeng, 2023-08-02:
Hi 
I found the this link:[linux.bin and rootfs.bin](https://github.com/niceboygithub/AqaraCameraHubfw/tree/main/modified/M2_2022)is no longer valid. where can I download these two files?

niceboygithub, 2023-08-06:
Because they are not released because M3 will be released soon.

tomtzeng, 2023-08-06:
What impact do release files from M2 POE have on the upcoming release of M3?

niceboygithub, 2023-08-06:
M3 may use the same design. In order to use the similar method to enable telnet on M3, and providing lumi patch the issue, I need keep this method unpublished 
.

niceboygithub, 2023-08-23:
> Hi, I'd like to know if there is a complex but working method for now. @niceboygithub 

Yes

ashottonoyan, 2023-11-06:
https://forum.automata.id/t/topic/542

kindansljd, 2023-11-18:
> M3 may use the same design. In order to use the similar method to enable telnet on M3, and providing lumi patch the issue, I need keep this method unpublished .

If i want to flash my M2 2022 recently,  is there any secret way to get both linux.bin and rootfs.bin? 
Or I should keep waiting until M3 been released?

niceboygithub, 2023-11-18:
There will be two gateways, M3 and another model, both of them may use the same method to update firmware in uboot.
In case of lumi change the method, I prefer to wait for a little while to release.

flyou, 2023-12-06:
waitting  for m2 2022 firmware

ashottonoyan, 2024-01-05:
M3 is out https://www.aqara.cn/Hub-m3_overview

closed 2024-01-24 by niceboygithub