niceboygithub/AqaraCameraHubfw #24: how modify roofs.bin?

sdavides, 2023-11-11:
Hi, can you explain to me how to change the firmware?
how modify roofs.bin?

I would like to modify last firmware aqara g3:
https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.camera.gwpgl1/20231030143851_lumi.camera.gwpgl1_AIOT_V4.0.6_0011.0004_20231027_6b956f.bin

I tried with binwalk and unsquashfs without success.

Thanks

niceboygithub, 2023-11-12:
The rootfs is created as squashfs, but it is signed.
The official boot sequence of G3 use signed rootfs. If not signed rootfs, it will boot failed.

sdavides, 2023-11-12:
thanks for the reply, can you explain how to do it?  I will publish the new updated version with hack of the g3 camera tested

sdavides, 2023-11-22:
I'm just missing the signature, could you help me?

Official rootfs.bin:
~ # /bin/fw_upgrade /mnt/sdcard/rootfs.bin
image magic:fwss, signature:lmrt, crc32:1922183475, len:25792768

My rootfs.bin:
/bin/fw_upgrade /mnt/sdcard/firmware.img
image magic:ºÁ¶X, signature:¸Qaz, crc32:998604687, len:1208766611
Error:image header magic is not fixed!fwcs, ºÁ¶X!
fw_upgrade error:-2

thanks

niceboygithub, 2023-11-22:
As I said, the firmware of G3 is signed rootfs. Only lumi have key to sign rootfs.

sdavides, 2023-11-22:
OK sorry, so it's not possible? 
As you did for the ones you modified here:
https://github.com/niceboygithub/AqaraCameraHubfw/tree/main/modified
?

niceboygithub, 2023-11-22:
E1 and M2 PoE do not use signed rootfs.

sdavides, 2023-11-22:
OK thankyou !!

closed 2023-11-22 by niceboygithub

sdavides, 2024-03-04:
I saw that you are making the modified g3 firmware, good job! If you like, can you tell me how you did it for signing the .bin?