niceboygithub/AqaraCameraHubfw #26: Can I update modified/G3/4.0.7_0006.0008?

mikeinblue, 2023-12-04:
Thanks for your great work.
I was wondering can I update modified/G3/4.0.7_0006.0008 firmware use sd card? Or I only can downgrade the firmware use sdcard to get rooted?(mine is 4.0.7 rightnow)
Thank you for yourtime.

mikeinblue, 2023-12-06:
Thanx for your answer.I will have a try

Necroneco, 2023-12-06:
But where is the g3_update.sh ?

mikeinblue, 2023-12-07:
> Can NOT use sdcard to update modified firmware.
> 
> You need use g3_update.sh to update modified firmware after downgrade firmware.

I downgraded to 3.3.4 now,but where I can find the g3_update,sh?

niceboygithub, 2023-12-07:
Will release soon.

olarodes, 2024-01-11:
@niceboygithub 
Thank you for your amazing work! With your guidance in another issue #27 , my G2Hpro was able to open telnet. The firmware version is still at 3.3.4_0014.0004. I hope it's okay to ask if there are any plans to provide a modified firmware for G2Hpro. Keep up the great work!

niceboygithub, 2024-01-11:
Create an executive post_init.sh in /data/scripts.

Then upgrade to latest fw.

The method of post_init.sh was not removed in G2H pro.

niceboygithub, 2024-01-11:
There is a risk to use g3_update.sh, do you want to take this risk?

The sdcard updated method can flash to original fw if g3_update.sh was failed to flash modified fw. After restored to original fw, go to try g3_update.sh again.

olarodes, 2024-01-12:
> Create an executive post_init.sh in /data/scripts.
> 
> Then upgrade to latest fw.
> 
> The method of post_init.sh was not removed in G2H pro.

Wow, thank you for the reply! So I just need to create the sh file with the following method

```
mkdir /data/scripts
cd /data/scripts
/data/bin/curl -s -k -L -o /data/scripts/post_init.sh https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/binutils/post_init.sh
chmod +x /data/scripts/post_init.sh
```

and then use the Aqara app for OTA? That sounds super easy, can't wait to try it out! Thank you! 🙌

closed 2024-01-16 by niceboygithub