niceboygithub/AqaraCameraHubfw #31: M2 POE Gateway open telnet  fail

miniknife88, 2024-01-25:
follow telnet readme,the LED does not turn purple

niceboygithub, 2024-01-25:
Format USB thumb with FAT32 in Windows.

miniknife88, 2024-01-25:
FAT32 format is ready，but the LED does not turn purple

niceboygithub, 2024-01-25:
Single partition format to FAT32.

The gateway must be power off before hold the button.

wu-cl, 2024-02-03:
I have the same problem. Use fat32 single partition usb device，power off before hold the button.
After re-Plug-in, the yellow light goes out and starts flashing blue, no purple.

Sircoco2, 2024-02-10:
the yellow light goes out and starts flashing blue, no purple.

niceboygithub, 2024-02-25:
> the yellow light goes out and starts flashing blue, no purple.

Please check the format of your USB Disk. (MBR, single partition with FAT32)

lihuohuo9527, 2024-03-02:
There will still be a TELNET password
Devices in aqara getaway cannot be used after M2 2022 (POE) restarts

lihuohuo9527, 2024-03-02:
1. There is still a telnet password, but aqara getaway is working properly

2. After restarting the m2 Poe, the devices in the aqara getaway cannot be used

yebjx, 2024-05-15:
telnet m2-2022-poe ，The username is admin, and the password is left blank, and the password is incorrect

vachenli, 2024-06-16:
my usb disk is mbr signle partition with FAT32,and the m2 2022 light is change to purple,but , i cant telenet after it restart

Chengxuyuan0, 2024-06-25:
> telnet m2-2022-poe，用户名为admin，密码留空，密码不正确
I tried to connect to the M2-2022-POE device using Telnet with the username 'admin' and left the password field blank, but the password was not accepted. I encountered the same issue.

xiaozhubin, 2024-10-08:
> > telnet m2-2022-poe，用户名为admin，密码留空，密码不正确
> > I tried to connect to the M2-2022-POE device using Telnet with the username 'admin' and left the password field blank, but the password was not accepted. I encountered the same issue.

试试用户名 root

yunjie-li, 2024-12-24:
试了很多次，现象还是从黄色的灯变成蓝色的灯。我检查了u盘格式确认是MBR FAT32. 文件也没问题。操作也没有问题，m2 固件版本是最新的4.3.1_0010.0013. 是这个固件不支持了吗。