niceboygithub/AqaraCameraHubfw #32: Aqara Camera G3 - Unable to update to latest FW

T-REX-XP, 2024-01-27:
**HW:**  lumi.camera.gwpagl01, model: AH_G3
**SW version:** 
**How to reproduce**:
```
cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/aqarahub/binutils/curl?viasf=1" && chmod a+x /tmp/curl
/tmp/curl -s -k -L -o /tmp/g3_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraCameraHubfw/main/modified/G3/g3_update.sh
chmod a+x /tmp/g3_update.sh && /tmp/g3_update.sh
```
**Logs:**
```

 chmod a+x /tmp/g3_update.sh && /tmp/g3_update.sh
psvr ok,wait=1
g3_update.sh revision: 2
type: lumi.camera.gwpagl01, model: AH_G3
platform: aiot, path: /tmp/fw.tar.gz, sign: 0, type: 1
killall: app_monitor.sh: no process killed
killall: ppcs: no process killed
killall: webrtc: no process killed
killall: rtsp: no process killed
killall: ai: no process killed
killall: recorder: no process killed
killall: uvc: no process killed
killall: vidicond: no process killed
dfu size start
Original OTA package : /tmp/fw.tar.gz
Unpack path : /data/ota_unpack
Firmware path : /data/ota_dir
OTA packages size(kb) : 16384
Available ROM size(kb): 84704
Available RAM size(kb): 14496
Update to 4.0.8_0010.0008
Get packages, please wait...
Update failed, reason: get packages failed!
```

**Error:**
/tmp/g3_update.sh: line 667: /bin/nandwrite: not found

niceboygithub, 2024-01-28:
Please upgrade 3.3.9 first. (remember that remember post_init.sh in /data/scripts to enable telnet)

niceboygithub, 2024-02-25:
Please update to the latest g3_udpate.sh and try again

sdavides, 2024-03-06:
with update 3.3.9 the problem is solved, but the reboot is back to 3.2

T-REX-XP, 2024-04-25:
solved for me with latest updates.

closed 2024-04-25 by T-REX-XP