niceboygithub/AqaraCameraHubfw #33: G3 update modfirmware

mikeinblue, 2024-02-03:
downgrade to 3.3.4 opened telnet
putty update with issue:
[image]

what can I do next? Thx a lot

wizardofozzie, 2024-02-05:
@mikeinblue I rewrote the guide here - https://github.com/wizardofozzie/aqara_g2hpro_hack
buy those guys a coffee if you get it working

mikeinblue, 2024-02-06:
> @mikeinblue I rewrote the guide [here](https://github.com/wizardofozzie/aqara_g2hpro_hack)- start at step 7 buy those guys a coffee if you get it working

I have already telneted.But I can't update modifyfirmware with telnet

niceboygithub, 2024-02-06:
Ignore the update of zigbee.
Just modify the lone 69 in g3_update.sh to COOR_MD5SUM=“”

mikeinblue, 2024-02-08:
> Ignore the update of zigbee. Just modify the lone 69 in g3_update.sh to COOR_MD5SUM=“”

Thanks for reply.I am noob in linux,so I just can type"**_cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/aqarahub/binutils/curl?viasf=1" && chmod a+x /tmp/curl
/tmp/curl -s -k -L -o /tmp/g3_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraCameraHubfw/main/modified/G3/g3_update.sh
chmod a+x /tmp/g3_update.sh && /tmp/g3_update.sh_**"    and don't know how to  modify the lone 69 in g3_update.sh to COOR_MD5SUM=“”.

BTW:Happy Chinese New Year

sdavides, 2024-03-06:
I try, but error:

===Update uboot===
===Update kernel===
=== KERNEL1 ====
main:102
Erasing 128 Kibyte @ 4e0000 -- 100 % complete
/tmp/g3_update.sh: line 714: /bin/nandwrite: not found
/tmp/g3_update.sh: line 715: /bin/nanddump: not found
main:102
Erasing 128 Kibyte @ 4e0000 -- 100 % complete
/tmp/g3_update.sh: line 714: /bin/nandwrite: not found
/tmp/g3_update.sh: line 715: /bin/nanddump: not found
cat: can't open '/tmp/kernel_head': No such file or directory
main:102
Erasing 128 Kibyte @ 4e0000 -- 100 % complete
/tmp/g3_update.sh: line 714: /bin/nandwrite: not found
/tmp/g3_update.sh: line 715: /bin/nanddump: not found
cat: can't open '/tmp/kernel_head': No such file or directory
main:102
Erasing 128 Kibyte @ 4e0000 -- 100 % complete
/tmp/g3_update.sh: line 714: /bin/nandwrite: not found
/tmp/g3_update.sh: line 715: /bin/nanddump: not found
cat: can't open '/tmp/kernel_head': No such file or directory
Update failed, reason: OTA failed!

[image]

sdavides, 2024-03-06:
update 3.3.9 problem solved

closed 2025-11-16 by niceboygithub