niceboygithub/AqaraCameraHubfw #35: G2H get bricked

redelva, 2024-02-21:
With guide of telnet, i enable telnet successful.

change the s90app script for auto enable rtsp and image grabber when restart g2h

with rtsp_start.sh in s90app and httpd add in script, after restart the camera, it can not boot up.

sounds like it can not boot up for added script.

Now it can not be reset and connected, any solution for this case?

redelva, 2024-02-21:
any solution for restore to stock firmware?

niceboygithub, 2024-02-25:
use the UART to save it

redelva, 2024-02-25:
any guide for uart to save?

i remember the init script, start httpd and block boot procdure

Worrah, 2025-05-13:
> any guide for uart to save?
> 
> i remember the init script, start httpd and block boot procdure
> […](#)
> 
>  use the UART to save it

hello mate. did you find a solution aside the uart? i've bricked mine as well =_=