niceboygithub/AqaraCameraHubfw #36: Aqara Hub E1 New firmware: AIOT_V4.1.0_0003.0010

Necroneco, 2024-03-14:
```json
"updateTime": 1710387835000,
"fileSize": 15564904,
"state": 3,
"releaseDate": 1710387835000,
"imageType": "bin",
"updateLog": "1.优化设备相关功能； 2.修复已知问题；",
"isNecessary": 0,
"firmwareMD5": "0791bf4f51cec98a425d38d2a2f19034",
"isForce": false,
"part": "main",
"firmwareCRC32": "CAC3AA0B",
"firmwareVersion": "4.1.0_0003.0010",
"createTime": 1709710618000,
"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.aqcn02/20240221150155_lumi.gateway.aqcn02_AIOT_V4.1.0_0003.0010_20240220_f19034.bin",
"upgradeTakesTime": 0
```

Thanks~~

niceboygithub, 2024-03-15:
It sometimes fail on the zigbee update. Try to manual update zigbee fw.

Necroneco, 2024-03-15:
Emm, how to manual update zigbee fw?

niceboygithub, 2024-03-17:
```
cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/aqarahub/binutils/curl?viasf=1" && chmod a+x /tmp/curl
/tmp/curl -s -k -L -o /tmp/coor.bin https://raw.githubusercontent.com/niceboygithub/AqaraCameraHubfw/main/original/E1/4.0.4_0006_4.1.0_0003/Network-Co-Processor.ota
md5sum /tmp/coor.bin
zigbee_msnger zgb_ota /tmp/coor.bin
```

Necroneco, 2024-03-17:
> ```
> cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/aqarahub/binutils/curl?viasf=1" && chmod a+x /tmp/curl
> /tmp/curl -s -k -L -o /tmp/coor.bin [...]

Success! Thank you!

closed 2024-03-17 by Necroneco