niceboygithub/AqaraCameraHubfw #37: M2PoE Upgrade Error

CharmingCheung, 2024-03-24:
My M2PoE is currently running on the modified firmware version 4.0.6_0011.0004. I am attempting to upgrade to version 4.1.0_0007.0010 via Telnet, but I have encountered an error. Can you please assist me with this issue?

``` log
psvr ok,wait=1
type: lumi.gateway.iragl8, model: AH_M2
platform: aiot, path: /tmp/fw.tar.gz, sign: 0
Original OTA package :
Unpack path : /data/ota_unpack
Firmware path : /data/ota_dir
OTA packages size(kb) : 16384
Available ROM size(kb): 50868
Available RAM size(kb): 15648
Update to 4.1.0_0007.0010
Get packages, please wait...
Got package done
env right
===Update IR-Controller===
===Update zigbee-coordinator===
zb_coordinator: 0021, coor_bin_name:Network-Co-Processor.ota
Network-Co-Processor_115200_MG21_0010_20231206_FA2882.ota
rm: can't remove '/data/ota-files/*': No such file or directory
==2==zb_platform: 0021, MG21
[INFO]: exec zgb_ota success  (command.c:374)
zigbee_msnger zgb_ota /data/ota-files/Network-Co-Processor.ota
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
[INFO]: exec get_zgb_ver failed  (command.c:374)
coordinator ota fail
Update failed, reason: OTA failed!
```

CharmingCheung, 2024-03-24:
I just forked the repository and attempted to change the firmware versions referenced in m2poe_update.sh to 4.0.8_0008.0010 and 4.0.6_0011.0004 respectively, before re-flashing via Telnet. 
The attempt to flash 4.0.8_0008.0010 failed, but 4.0.6_0011.0004 was successful. 
Upon reviewing the commits for 4.0.8_0008.0010, I noticed some changes to the Zigbee-related code. Could the issue be related to this? 
If I want to flash the 4.1.0xxx version again, do I need to OTA to the official 4.1.0 version before flashing the modified firmware?

sxzcy, 2024-09-09:
我也遇到了这个错误。我的解决方案是：重新U盘刷回4.04，再telnet升级至4.18