niceboygithub/AqaraCameraHubfw #39: aqara g3 doesn't scan qr code

Guizzan, 2024-06-07:
I have upgraded the software to version 3.3.4. Reseted the camera and pressed for 10 seconds to activate pairing mode, but for some reason it won't read any QR code (aQRootg3.py generated or original QR code from Aqara home app). When I update the software back to the latest version, it starts reading again. What can I do in this situation?

sdavides, 2024-06-19:
strange, not even the original app?

did you copy all the files to downgrade root.fs and others?

Guizzan, 2024-06-20:
Yes, I first tried with only root.fs, then with all the files.

EricCorleone, 2024-08-07:
Same issue here @niceboygithub

EricCorleone, 2024-08-08:
Is it possible that it's because there are blank space in my ssid? But it can't explain why g3 couldn't read the qr code from aqara app...

Guizzan, 2024-08-08:
> Is it possible that it's because there are blank space in my ssid? But it can't explain why g3 couldn't read the qr code from aqara app...
> 
> I tested v3.2.7 tonight, and it doesn't work either.

My ssid or password doesn't contain any blank space or invalid characters. It's most probably some software/hardware lock that prevents to use lower versions.

EricCorleone, 2024-08-08:
> My ssid or password doesn't contain any blank space or invalid characters. It's most probably some software/hardware lock that prevents to use lower versions.

I searched for solutions on the Internet, and it seems this method still worked in 2023, but my camera is also produced in 2023, so it doesn't look like a hardware lock.

[image]
I also tried to update it to v4.1.6, paired it with Aqara Home app, then downgraded it to v3.3.4 without resetting the camera. As you can see, after that the camera is still connected to the Aqara Home app and the version number is 3.3.4, so the firmware is successfully flashed into the camera, therefore, it doesn't look like a software lock either. But somehow it just cannot read any QR code anymore. So weird!

Pitti3301, 2024-10-21:
Same issue here, the camera won't read any QR code. Is there another way to activate telnet on a downgraded g3?

sdavides, 2024-11-16:
1. the qrcode image must not be too large, but small.
2. the wifi sid and password must not have special characters.
3. try modifying the script, I also had problems and tried changing the payload if I remember correctly.

tomaszbartoszewskitb, 2024-12-06:
Having the same issue on a G3 shipped with 4.0.2 firmware. When downgrading to 3.3.4, 3.3.2 or 3.2.7 the camera cannot read QR code from both the app and the script at different sizes and distances from the camera. Before downgrading the camera was reset with 10 clicks. I have also tried flashing rootfs.bin alone, and all of the .bin files.

Pitti3301, 2024-12-07:
With the conditions from sdavides, the camera is connected to the WLAN for a few seconds. Then the error message "account binding failed" appears. What could be the reason for this?

sdavides, 2024-12-07:
Is correct as written in the guide.
"Then the error message "account binding failed" appears." it means that it ran the qr script but did not bind with aqara, now all that remains is to restart the scanning procedure, without restarting device, with official qr , if I'm not mistaken, hold down the button and the voice indicates to scan

Pitti3301, 2024-12-07:
That worked great. Thanks for the tip.

sdavides, 2024-12-07:
> That worked great. Thanks for the tip.

See my repo for Aqara G3 !!

Pitti3301, 2024-12-08:
Another question about the update. Version 3.3.4 is currently running. The next step is to install version 3.3.9 manually before installing a custom version. What steps are necessary to update to 3.3.9?

sdavides, 2024-12-08:
Download 3.3.9
Unzip
Copy file on sd
Hold down button and start device
Follow guide on my repo AqaraPost

Pitti3301, 2024-12-08:
I unpacked the bin file. As a result I have the following files: ControlBridge.bin ir_ota_cm_v12.bin IRController.bin linux.bin and rootfs.bin. Which file(s) is now required on the SD card?

sdavides, 2024-12-08:
all file

sdavides, 2024-12-08:
rereading the repo, an update has been made so it is not mandatory to move to 3.3.9

[Add support of upgrade Aqara Camera G3 from 3.3.4]

[image]

tkboisvert, 2025-01-10:
I'm having the same issue. Aqara G3 Purchased in USA off of Amazon. I can change firmware versions just fine, but when I'm on lower FW versions I cannot use any QR codes to config the G3.

when in a lower FW version, my G3 cannot scan QR codes from either the official Aqara App or Wh1terat's QR tool. The only way I can connect to G3 when in v3.3.4 is via homekit ( that's how I've confirmed the FW is flashing fine ).

If I flash back up to a more recent version I can use the Aqara app again.

My Process is same as others are saying, flash new FW, click 10 times, attempt to scan QR code on PC with varying sizes and brightness. Anyone solved this?