niceboygithub/AqaraCameraHubfw #40: How to update the modified firmware?

EricCorleone, 2024-07-11:
If I have already flashed the modified firmware for my hub E1, and if I want to update the modified firmware, should I just run the following command again?
```
cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/aqarahub/binutils/curl?viasf=1" && chmod a+x /tmp/curl
/tmp/curl -s -k -L -o /tmp/e1_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraCameraHubfw/main/modified/E1/e1_update.sh
chmod a+x /tmp/e1_update.sh && /tmp/e1_update.sh
```
Will this behavior erase my devices connected to the hub?
Just want an answer before I try, thanks!

Necroneco, 2024-07-14:
Yes, run it again. don't worry

EricCorleone, 2024-07-14:
OK, thanks!

closed 2024-11-03 by niceboygithub