niceboygithub/AqaraCameraHubfw #41: M2PoE upgrade to 4.1.6 modified firmware failed

eggleader-zhang, 2024-07-19:
I hava 3 M2PoE devices.
But 2 of them can not upgrade to 4.1.6
the logs show:
``` 
platform: aiot, path: /tmp/fw.tar.gz, sign: 0

Original OTA package : 
Unpack path : /data/ota_unpack
Firmware path : /data/ota_dir
OTA packages size(kb) : 16384
Available ROM size(kb): 50932
Available RAM size(kb): 17616
Update to 4.1.6_0018.0012
Get packages, please wait...

Got package done

env right
===Update IR-Controller===
===Update zigbee-coordinator===
zb_coordinator: 0021, coor_bin_name:Network-Co-Processor.ota
Network-Co-Processor_115200_MG21_0012_20240523_9E975B.ota

rm: can't remove '/data/ota-files/*': No such file or directory
==2==zb_platform: 0021, MG21

[INFO]: exec zgb_ota success  (command.c:374)
zigbee_msnger zgb_ota /data/ota-files/Network-Co-Processor.ota

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

eggleader-zhang, 2024-07-19:
finally it shows ota failed
``` 
[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)

[INFO]: exec get_zgb_ver failed  (command.c:374)
coordinator ota fail
Update failed, reason: OTA failed!

sxzcy, 2024-09-09:
我也遇到了这个错误。我的解决方案是：重新U盘刷回4.04，再telnet升级至4.18

eggleader-zhang, 2024-09-09:
> 我也遇到了这个错误。我的解决方案是：重新U盘刷回4.04，再telnet升级至4.18

我这边手动export了代理地址，就没这个问题了，应该是下载OTA文件的时候，遇到了网络问题导致的。或者把M2的网关指向有Clash的路由器也行

closed 2025-11-16 by niceboygithub