niceboygithub/AqaraCameraHubfw #44: Aqara G2H enable telnet

thankarezos, 2024-08-14:
I have chinese version of the Aqara g2h. I am trying to enable telnet but nothing happens, When i try to telnet 192.168.1.28 I get connection refused. I also tried to connect it to the aquara home app but it doesnt connect. only to the mija app. the firmware is 3.5.8. Also I have a CH340G usb module. if flashing like that is not possible to this version I would like somone to point me on a good guide on what should i solder where to flash it via hardware.
[image]

[image]

hugodias-git, 2024-08-20:
Hello, I have the same camera, G2 ZNSXJ12LM with Firmware 3.5.8_0062.0356 and I cannot achieve to open Telnet.
I've migrated my xiaomi devices to HA and the camera makes hub for several sensors.

Does anyone can help to explain how to open telenet on the camera?

Many thanks

hugodias-git, 2024-08-20:
I think my G2 ZNSXJ12LM is not acepting the hostname file, perhaps the firmaware version is not compatible. Just wanted to know if there is a solution, even with UART (solding) or not. I'm getting confused with the different versions and models (G2 and G2H).   I really appreciate this repository from @niceboygithub but I'm getting some issues to solve my camera integration.

Thanks you all

niceboygithub, 2024-08-20:
The hostname method works on G2 and G2H.

thankarezos, 2024-08-21:
> Hello, I have the same camera, G2 ZNSXJ12LM with Firmware 3.5.8_0062.0356 and I cannot achieve to open Telnet. I've migrated my xiaomi devices to HA and the camera makes hub for several sensors.
> 
> Does anyone can help to explain how to open telenet on the camera?
> 
> Many thanks

I suggest using a sonoff bridge for zigbee. More reliable and very cheap (around 15$). I flashed mine with tasmota, although I think it's not required to flash it any more, there is a sonoff integration (i think) although flashing the sonoff bridge is very easy. You just plug a couple of cables on the holes (not even need soldering) and you flash it with CH340G usb (costs around 3$), the reason I want to integrate the xiaomi camera is solely for the camera function and not for the zigbee.

thankarezos, 2024-08-21:
> The hostname method works on G2 and G2H.

Apparently it doesn't work on newer firmwares or maybe specific models. I have a Chinese version of the camera with the Chinese firmware (it even speaks Chinese). Either this or it got patched at some point. I'll have a friend of mine that works with hardware solder UART cables (because apparently I sucked and the pads being too small didn't help either) and try that way. But the hostname method did not work.

hugodias-git, 2024-08-21:
@thankarezos did you reached to open telnet by UART with your friend? I would like to try it before buy another ZigBee gateway.
Thanks

thankarezos, 2024-08-22:
> @thankarezos did you reached to open telnet by UART with your friend? I would like to try it before buy another ZigBee gateway. Thanks

not yet. I am gonna talk to him soon. But I still suggest finding a better alternative for ZigBee because even if you manage, I don't think it's a very reliable ZigBee gateway, and may not integrate well with home assistant. And even if UART works, you still going to need a usb to uart to connect to the serial, and you need someone with some skill to solder the cables because from experience the pads are tiny and so close together so you can't just pour some solder and hope for the best.   https://digiblur.com/2020/07/25/how-to-use-the-sonoff-zigbee-bridge-with-home-assistant-tasmota/ this is what I did for ZigBee bridge and costed me less than 20$. Check for updates tho because it might not need flashing anymore

hugodias-git, 2024-09-15:
@niceboygithub any idea what cpuld be or how to find the error ?

niceboygithub, 2024-09-16:
Camera G2 is NOT in the supported list of the AqaraGateway

hugodias-git, 2024-09-16:
Ohhh, but in the description it says that Aqara Camera Hub G2/G2H (ZNSXJ12LM) is supported.
My camera is G2 ZNSXJ12LM (chinese version).

@niceboygithub so finally is not supported?

I reached to open Telnet by UART and I installed the custom [Not Flash Custom firmware method (for G2H)](https://github.com/niceboygithub/AqaraGateway/tree/72561429919dcb542e9d2e065a07c38b076ee4e8#not-flash-custom-firmware-method-for-g2h) Mosquitto binaries to open mqtt.
So for my camera this won't work? You never reached to unblock it? Is there so difference between the chinese version and worlwide for ZNSXJ12LM ?

Thank you for your help

niceboygithub, 2024-09-17:
This repository is NOT AqaraGateway. It is AqaraCameraHubfw.

hugodias-git, 2024-09-17:
Sorry, you are wright.

Thank you

closed 2025-02-27 by niceboygithub