niceboygithub/AqaraCameraHubfw #46: Aqara G2 - Streaming on HA

hugodias-git, 2024-09-17:
Hello,
I've 3 camera G2 ZNSXJ12LM (chinese version) that I would like to add to my HA Environement.
I've understood that unfortunately this G2 chinese version don'y have mqtt so it's not possible to use the gateway, but for video streaming the rstp support can be used.
I've some questions:
- How can I install it and make it available on the camera.
- How to reach it via HA
- If I have to install some binaries or flash something, I will loose connection to Mi Home app that I have currently?

Thank for your help, I'm stuck at this point.

Hugo

hugodias-git, 2024-09-24:
@niceboygithub do you know to get camera imagem/streaming to HA ?
Thanks

niceboygithub, 2024-09-24:
I did. But I forgot how.

G2 is a too too too old camera. And the software design is bad. Please throw it in the trash can.

hugodias-git, 2024-09-25:
They are good, and work well with MiHome.
But it if complicated to put it work with HA this are not good news.

niceboygithub, 2024-09-25:
The rootfs of G2 can be erased after rooted. Do you still think G2 is good?

hugodias-git, 2024-09-26:
I said it's good in image quality and design/size.
In terms of software the fact that is not simple to integrate as the followers G2H and G2 PRO it's not good at all.

Thanks anyway.