niceboygithub/AqaraCameraHubfw #47: G3 3.3.9 trying to upgrade to latest formare

brentgl, 2024-09-26:
Hi
I am running the script on an Aqara G3 on firmware 3.3.9 and keep getting the following error.  It doesnt affect the G3 - it still stays on 3.3.9 and telnet still works.  Is there an advantage of going to latest firmware? or a way around this error?

psvr ok,wait=1
g3_update.sh revision: 2
type: lumi.camera.gwpagl01, model: AH_G3
platform: aiot, path: /tmp/fw.tar.gz, sign: 0, type: 1
dfu size start
Original OTA package : /tmp/fw.tar.gz
Unpack path : /data/ota_unpack
Firmware path : /data/ota_dir
OTA packages size(b) : 28201064
Available ROM size(b): 86683648
Available RAM size(b): 59564032
Update to 4.1.8_0018.0013
Get packages, please wait...
Got package done
===Update IR-Controller===
===Update zigbee-coordinator===
start recv
[INFO]: exec zgb_ota success  (command.c:374)
start recv
[INFO]: exec get_zgb_ver failed  (command.c:374)
start recv
[INFO]: exec zgb_ota success  (command.c:374)
start recv
[INFO]: exec get_zgb_ver failed  (command.c:374)
start recv
[INFO]: exec zgb_ota success  (command.c:374)
start recv
[INFO]: exec get_zgb_ver failed  (command.c:374)
Update failed, reason: OTA failed!

Thanks
Brent

niceboygithub, 2024-09-26:
Manually update zigbee fw first

brentgl, 2024-09-26:
any pointers on how I do that?
Thanks

brentgl, 2024-09-26:
I tried

```
cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/aqarahub/binutils/curl?viasf=1" && chmod a+x /tmp/curl

/tmp/curl -s -k -L -o /tmp/coor.bin https://raw.githubusercontent.com/niceboygithub/AqaraCameraHubfw/main/original/G3/4.1.8_0018.0013/ControlBridge.bin
md5sum /tmp/coor.bin
zigbee_msnger zgb_ota /tmp/coor.bin
```
but nothing - where is the file I need to update it?

Thanks Brent

brentgl, 2024-09-27:
done.
I ran the above scripts and removed the check within the G3setup.sh after downloading - all upgraded.
Brent