niceboygithub/AqaraCameraHubfw #48: Camera Hub G3 EU not downgrading

fifammlaf, 2024-10-27:
Hello,
I have a Camera Hub G3 EU that has a version that patched [the aQroot exploit](https://github.com/Wh1terat/aQRootG3)
when i tried to downgrade to firmware [3.3.4](https://github.com/niceboygithub/AqaraCameraHubfw/blob/main/stock/G3/20211216200913_lumi.camera.gwpagl01_AIOT_V3.3.4_0007.0004_20211216_28f9c3.bin) with the SD card Method it didn't work could the V3.3.4 firmware be exclusively for the Chinese version?

niceboygithub, 2024-11-03:
It shall works. Please check the format of sdcard.

fifammlaf, 2024-11-16:
I verified the format of the sdcard and it is FAT32 the only firmware from the ones i tried that flashed was [4.0.7](https://github.com/niceboygithub/AqaraCameraHubfw/blob/main/original/G3/4.0.7_0006.0008/rootfs_4.0.7_0006.0008.bin)