niceboygithub/AqaraCameraHubfw #49: RTSP stop working on 4.1.9_0026.0013

sdavides, 2024-11-05:
before version 4.1.9_0026.0013 I used the rtsp url for the homeassistant webrtc video.

It seems like there is now a probe that restarts the process when you try to access url rtsp//, it keeps changing your username and password.

If I kill rtsp and restart it without "-a" to remove authentication, the process restarts with "-a".

I see a new process that I didn't notice before called "webrtc".

Can you help me?

sdavides, 2024-11-05:
when try url rtsp, the process stopped with: 
StreamParser internal error (0 + 800000 > 320000)

niceboygithub, 2024-11-06:
I think it is a bug of 4.1.9. 
1. downgrade to 4.1.8
2. Copy the rstp of 4.1.8 to 4.1.9
3. Wait next version release

sdavides, 2024-11-06:
Thankyou, how copy rtsp?

 File system is only read

niceboygithub, 2024-11-06:
I updated the rootfs. Please update again.

sdavides, 2024-11-06:
wow it works now, you are the best.

I would really like to know how you modify a signed squashfs.

Have you by any chance seen my nodered add-on for G3 with only http requests?

A thousand thanks

closed 2024-11-12 by sdavides