niceboygithub/AqaraCameraHubfw #52: E1 not enough storage to update

yttuan107, 2025-01-15:
after use commond to update, it's show "Not enough storage available", could u help me how to fix it, thanks a lot

[image]

rwalkow, 2025-01-22:
try to use the command:
`rm -rf /data/ota_dir/`

closed 2025-02-27 by niceboygithub

yttuan107, 2025-03-06:
thank ur help, but i type the command,it's just show the message,could you help me how can i do,thank a lot

[image]

rwalkow, 2025-03-10:
strange, try a different way:
```
chmod 777 /data/ota_dir/lumi_fw.tar
mv /data/ota_dir/lumi_fw.tar  /data/ota_dir/lumi_fw.tar.del
cat /dev/null >  /data/ota_dir/lumi_fw.tar.del
```
this should rename the file /data/ota_dir/lumi_fw.tar  to /data/ota_dir/lumi_fw.tar.del and resize it to 0 bytes, which should free up memory on the flash.

yttuan107, 2025-03-10:
Thanks a lot for your help, maybe something wrong or problem in my device, i still can't use this command. Sorry to bother you for so long. maybe i should give up to update.

[image]

rwalkow, 2025-03-12:
check if you have the right version of busybox
[image]

next try this:
/bin/busybox rm -rf /data/ota_dir/*

rwalkow, 2025-03-12:
lack space between /bin/busybox rm and -rf /data/ota_dir/*