niceboygithub/AqaraCameraHubfw #63: G2HPro firmware

maxromanovsky, 2025-07-20:
Please clarify if G2HPro firmware is supported?

In comparison with all the other cameras, G2HPro folder contains only a single script: `modified/G2HPro/g2hpro_update.sh`

And this script references files that are not found in the repository, for instance `https://raw.githubusercontent.com/niceboygithub/AqaraCameraHubfw/main/original/G2HPro/4.3.2_0024.0013/ControlBridge.bin`

niceboygithub, 2025-09-02:
The people who use G2H Pro is not too much and no one want to update custom firmware .

maxromanovsky, 2025-09-02:
@niceboygithub if you want to build it, I'll happily test it out

maxromanovsky, 2025-11-16:
@niceboygithub  I just saw that you've added a firmware. Thanks, I'll check it out in a few weeks (now is really busy time for me :D )