niceboygithub/AqaraCameraHubfw #65: aQrootG3

Wh1terat, 2025-09-02:
Hey,

You may remember me from a while ago, I discovered the qrcode exploit in the G3.
Well it seems those nice people at Aqara introduced a new flaw in the QRCode handling at some point in 4.x  😆 

My G3 refuses to update past 4.1.4 (despite 4.3.6 being available) and works in that, suspect it works in later versions and likely other models.

Before I update my repo, wanted to see if I can reach out to you and do some testing, perhaps on other models too?

If so, let me know how to reach you 👍

niceboygithub, 2025-09-02:
I know you forever.

There are many problems in Aqara Camera G3. I have four G3 cameras and got the following problems for around three years.
1. The rotation of ptz is broken.
2. The charger port is melt.
3. Sometimes it can not connect to WiFi after reboot.
4. The upgrade firmware was pulled by lumi cause issues frequently.

As I known, the version 4.3.6 was removed in Aqara App.
And what kind of test you want to test?

niceboygithub, 2025-09-02:
What is the logs while update?

I got such issue that reboot while updating before. It was cause the ram is ran out. After reboot, the update can be finished.

If you only add WiFi and telnet, it may be not enough. You need to check the watchdog.

The method to enable telnet is the same in every lumi products. Just press button 5-2-2-2-2-2.

I have no idea that get the did, key, mac without telnet.

niceboygithub, 2025-09-03:
The old method to generate password of root is using generate_pswd.sh with "did, mac and key".
The new method for new products is using generate_pswd.sh with "did, mac and dac_crt_md5", like G5Pro, G410, Touchscreen Dial V1 etc.

I think or guess that to generat the password of root is get the "did, mac, key or dac_crt_md5" from iOS/Andorid app. But due to my limitated knowledge, I can not find the way from disassemble Android apk.

Wh1terat, 2025-09-03:
Hmm so i grabbed a few firmwares:
lumi.camera.acn005 - V4.0.1
lumi.camera.acn017 - V4.3.3
lumi.camera.agl003 - V4.3.4

All seem to use the old pswd generation? does it start on a specific version? or model?

Also a cursory glance seems to show that they aren't vulnerable to the bug that was introduced into the g3 weirdly :/ not checked g2 etc.

There does seem like there might be a few other avenues to investigate for these devices though.

Just a little surprised that they differ so much, hate to think what code management at aqara is like.

niceboygithub, 2025-09-03:
>I remember when looking at the old method that there was no way to get all 3 from any of the Aqara endpoints used by the app so I don't think the app route will gain results unfortunately.

The developer of lumi shall be have a way to get these 3 properties. Maybe from the app or the cloud.

If the did of the product is start as 'lumi3", it use the new method.

Most lumi products have a key, but not all of them have the sdcard.

[image]

GalaxyGiant, 2025-12-03:
Can I know the account and password for pure telnet, which cannot be changed by firmware or QR code?
I'm only able to connect to Telnet via the physical button. I have an OEM version that doesn't have an SD card slot.